CTCaer/hekate #984: Black screen on hekate launch (booting switch regurarly)

Samybigio, 2024-06-19:
as soon as i updated hekate to latest version, upon entering it, i encountered an issue where the backlight does turn on (it bounces in brightness like it's booted hekate) but there is nothing displayed on the screen, just pitch black. i have a mariko v2 switch and despite saying compatible with mariko, it strangely doesn't work. for instance my friend's erista just works fine with latest hekate. i don't have latest atmosphere but i don't think i need that to at least boot into hekate. any help would be appreciated.

urherenow, 2024-06-19:
You DO need the latest Atmosphere. Hekate is a bootloader, not a CFW.

Samybigio, 2024-06-19:
welp i updated and it did work thanks! lmao

closed 2024-06-19 by Samybigio

urherenow, 2024-06-19:
I think it was mistyped... like maybe OP knows the gui is called nyx, and thinks booting into HOS is hekate 🤷‍♂️

That's the only thing that makes sense to me. But TBH, I was pretty drunk last night and didn't read the issue title correctly myself. Happy mistake that I helped?