CTCaer/hekate #986: Some games not launching after updating

HDD-Neptunia, 2024-06-21:
I updated my switch, (atmosphere, hekate, patches. Even fusee) to discover that some of my previous working games have stopped booting.

It feels I've tried everything, as I've tried downgrading, different sigpatches, maintainence mode. I've even put my rom on my partner's switch and it works just fine. I even removed things like mission control and reinstalling the game. Most other games boot without issue, but for the games in question, I just get 'An error occured' and it returns me to the home screen. Nothing more, no error code or anything

This only occurred after updating hekate to it's latest version to allow L4T Switchroot to boot, but now I can't boot into specific games such as Assassin's creed rogue or megadimension neptunia.  

I wouldn't usually leave an issue, as I can usually sort things myself. I've dealt with corrupted games, but I cannot for the life of me figure this out.

I hope I'm just being stupid and it's an easy fix, but I'd also like to know if anyone else can boot either of those two games on the latest software releases.

urherenow, 2024-06-21:
GitHub isn’t a support forum, and Hekate doesn’t load games. It’s a bootloader. The issue is likely piracy related, so you’re definitely can’t expect help from here.

HDD-Neptunia, 2024-06-21:
> GitHub isn’t a support forum, and Hekate doesn’t load games. It’s a bootloader. The issue is likely piracy related, so you definitely can’t expect help from here.

Thank you for the quick response. Can you point me in the right direction? Coincidentally, it's games that I actually own that aren't working. I only left an issue here because I don't rightly know where would be appropriate, considering that updating hekate is what broke my games.

Sorry for any ignorance, I'm not all too clued up on this. Thank you again

urherenow, 2024-06-21:
Gbatemp. Or discord. Or read th documentation yourself. If you update hekate, you should update atmosphere at the same time.

HDD-Neptunia, 2024-06-21:
To.be honest, I think that's where I messed up.

If I remember correctly, I updated hekate first, followed by atmosphere.
Thank you so much for the assistance. If I can't sort it, I'll probably just do a fresh install

CTCaer, 2024-06-21:
Don't pirate and they will magically work.

closed 2024-06-21 by CTCaer