CTCaer/hekate #987: Autoboot "OFF" on Hekate v6.20 is workless.

IntFinder706, 2024-06-23:
After updating the Atmosphere and Hekate (v6.20) to the latest, the setting of Autoboot to OFF cannot work.
Reboot each time the console will have the system go into the operation system running for the last time. 

You have to completely turn off the console and use the injection tool to turn the console on each time to enter into the Hekate menu. 
Previously, without the injection tool, I could enter the Hekate menu directly after rebooting through the Atmosphere tool.
I have set the Autoboot to OFF mode, but the reboot with Hekate processing without luck.

I hope this issue is not there and can be addressed by some simple settings. 

Thanks

urherenow, 2024-06-23:
You mean 0, right? Not OFF?

What's your updater2p setting?

CTCaer, 2024-06-23:
Whatever you used to reboot into a payload modified the reboot_payload.bin and set a boot entry override which it shouldn't never have done.
Software must never write to update.bin or reboot_payload.bin boot storage to select options.
They are supposed to read them in memory and edit them and do the warmboot exploit.
(Same deal for mariko, they should use r2c).

Anyway, replace both with a clean hekate

closed 2024-06-23 by CTCaer

IntFinder706, 2024-06-23:
> Whatever you used to reboot into a payload modified the reboot_payload.bin and set a boot entry override which it shouldn't never have done. Software must never write to update.bin or reboot_payload.bin boot storage to select options. They are supposed to read them in memory and edit them and do the warmboot exploit. (Same deal for mariko, they should use r2c).
> 
> Anyway, replace both with a clean hekate
> 
> EDIT: As pointed out, all that assumed you don't have fusee as reboot_payload.bin, which a copy of atmosphere replaces. And that's why update reboot 2 payload option exists in options.

Thanks for your clear explanation. This did get my problem addressed. 

Have a great day!

IntFinder706, 2024-06-23:
> You mean 0, right? Not OFF?
> 
> What's your updater2p setting?

Yes, I meant 0 (with .ini file under the bootloader folder), and also set the Autoboot in Hekate tool. 
Appreciate your help and suggestion, simply setting the updater2p can have the problem addressed. 

Have a great day!