CTCaer/hekate #992: Nyx is missing

gigelde, 2024-07-02:
an i also get a FatFS Error 13 and to make sure a FAT partition exists and Error: NOFAT

urherenow, 2024-07-02:
Likely a hardware issue with your reader or sd. Post a screenshot of your sd info screen.

gigelde, 2024-07-02:
[image]

gigelde, 2024-07-02:
> Likely a hardware issue with your reader or sd. Post a screenshot of your sd info screen.

[image]

gigelde, 2024-07-02:
> Likely a hardware issue with your reader or sd. Post a screenshot of your sd info screen.

I have to say that everything worked perfect till I updated the ofw via daybreak and choose fat option if I remember correctly I know it was the first one without the +

urherenow, 2024-07-03:
Have you tried looking at the card with your PC and an adapter? Seems pretty straightforward to me that your card is wiped out. Format it to fat32 on your PC. If you get nyx to load after that, format it again through the nyx partition menu.

gigelde, 2024-07-03:
> Have you tried looking at the card with your PC and an adapter? Seems pretty straightforward to me that your card is wiped out. Format it to fat32 on your PC. If you get nyx to load after that, format it again through the nyx partition menu.

Doesn't do anything

gigelde, 2024-07-03:
> Have you tried looking at the card with your PC and an adapter? Seems pretty straightforward to me that your card is wiped out. Format it to fat32 on your PC. If you get nyx to load after that, format it again through the nyx partition menu.

I have tried with fat expat fat 32

urherenow, 2024-07-03:
What doesn't do anything? Can you format it on your PC, and are you able to extract hekate and Atmosphere to the card? Or use https://www.sdcard.org/downloads/formatter/ then put hekate and atmosphere on it, then when/if nyx loads, format again within nyx.

This isn't a help forum, and "nyx missing" is not a bug, so you should ask for help in a forum on discord. Although I can't immediately see how to help you if you can't figure it out on your own. Many people would tell you to simply try a different sd card.

impeeza, 2024-07-03:
@gigelde this is not a support forum, please go to GBATemp or Discord,  by example on:  https://gbatemp.net/threads/switch-noob-paradise-ask-questions-here.488277/post-10397847

you will find:

It's almost sure your card is fake/dying, so backup all files on your SD card on your pc, format it to fat 32 using fat32 format (https://gbatemp.net/download/gui-format.33869/) then test your sd card using H2Testw (https://www.heise.de/download/product/h2testw-50539)

gigelde, 2024-07-03:
> @gigelde this is not a support forum, please go to GBATemp or Discord,  by example on:  https://gbatemp.net/threads/switch-noob-paradise-ask-questions-here.488277/post-10397847
>
>
>
> you will find: [...]

Thank you it worked

closed 2024-07-03 by gigelde