CTCaer/hekate #995: Keep having this show up when i inject hekate and it says nyx is missing

Blinkifyourok, 2024-07-11:
[image]
[image]

urherenow, 2024-07-12:
And it also says to update your bootloader folder. So do that.

Blinkifyourok, 2024-07-12:
How do I do that

urherenow, 2024-07-12:
Download the latest release. Grab the bootloader from the .zip, and drop it into your sd card. Overwrite all files, if asked.

In short: RTFM.

Blinkifyourok, 2024-07-12:
I've done that though I just keeps saying it

impeeza, 2024-07-12:
@Blinkifyourok this is not a support channel is only to report REAL issues on the code.

please search for help on Discord or GBATemp.

This problem is more like you have a fake/damaged sd card.

CTCaer, 2024-07-22:
extract the zip in the sd root and not a subfolder, and also to the correct partition if more than 1 fat partitions exist for some reason

closed 2024-07-22 by CTCaer