CTCaer/hekate #999: [BUG] Hekate 6.2.1 glitches in my 2017 switch and it's unusable

FerdinandoPH, 2024-07-21:
I have a 2017 switch (FW 17.0), and I've been using hekate with it for years. However, after updating to the latest version of hekate (6.2.1), I can't get it to run hekate anymore. All UI elements of the UI menu are glitched, and there is no way to interact with the menu (neither the screen nor the buttons do anything).
[image]

 I've tried:

1. Deleting my bootloader folder and copy all files from github
2. Checking for corruption in my SD and my RCM loader (using the Windows error detector) 
3. Using TegraRCM instead of my RCM loader

Yet I still get the same problem.
What can I do?

FerdinandoPH, 2024-07-21:
Update: The left joystick seems to work, but only makes corruption worse
https://youtu.be/XI5Hq6dBPic

impeeza, 2024-07-21:
Hello there, Is most likely your SD Card is becoming corrupted (they die with the time), this is not a support site, please go to GBATemp or Discord for support.

Try the next:

- Backup all your card,
- Download SD Formatter https://www.sdcard.org/downloads/formatter/ and format your SD Card,
- Then download H2TestW https://h2testw.org/ and test all surface of your card.
- if the card is ok, copy only Hekate to your card, and boot in to NYX and reformat the card and start over.

Si sigues teniendo problemas yo soy Impeeza en GBATemp.net.

urherenow, 2024-07-21:
Doesn't look like an SD issue to me. Does look like a hardware issue though. Might need to reflow your ram and apu. But sure, why not... try a new sd first with nothing other than atmosphere and hekate on it, just to be sure. Maybe even try without nyx.bin to see if you can navigate the other method with the volume and power buttons.

FerdinandoPH, 2024-07-21:
> Hello there, Is most likely your SD Card is becoming corrupted (they die with the time), this is not a support site, please go to GBATemp or Discord for support.
>
> Try the next:
>
> * Backup all your [...]

After backing up my SD, formatting it and testing it with h2testw, I still have the same problem. The UI is glitched and it won't respond. However, removing nyx lets me use hekate without the UI and it works fine.
On the one hand, it might be a hardware issue, just like @urherenow said. On the other hand though, I tried hekate 6.2.0 (and nyx 1.6.2), and although the UI elements have a glitchy appearance, the touchscreen works again and I can use hekate with nyx, so maybe it has something to do with the latest version of nyx?

impeeza, 2024-07-21:
> > Hello there, Is most likely your SD Card is becoming corrupted (they die with the time), this is not a support site, please go to GBATemp or Discord for support.
> > Try the next:
> > 
> > * Backup all your card,
> > * Download SD Formatter https://www.sdcard.org/downloads/formatter/ and format your SD Card,
> > * Then download H2TestW https://h2testw.org/ and test all surface of your card.
> > * if the card is ok, copy only Hekate to your card, and boot in to NYX and reformat the card and start over.
> > 
> > Si sigues teniendo problemas yo soy Impeeza en GBATemp.net.
> 
> After backing up my SD, formatting it and testing it with h2testw (which said that my SD is ok), I still have the same problem. The UI is glitched and it won't respond. However, removing nyx.bin lets me use hekate without the UI and it works fine. On the one hand, it might be a hardware issue, just like @urherenow said. On the other hand though, I tried hekate 6.2.0 (and nyx 1.6.2), and although the UI elements have a glitchy appearance, the touchscreen works again and I can use hekate with nyx, so maybe it has something to do with the latest version of nyx?

Sadly that is a confirmation you have a Hardware issue,  take the console to a repair shop to reball the APU and memory RAM chips.  Sorry.

CTCaer, 2024-07-22:
edit `sd:/bootloader/nyx.ini` and change `bpmpclock=1` to `bpmpclock=2` or 3.
and btw when you update hekate you are not supposed to delete bootloader folder and your settings

closed 2024-07-22 by CTCaer

clemchataigner, 2025-04-27:
> edit `sd:/bootloader/nyx.ini` and change `bpmpclock=1` to `bpmpclock=2` or 3. and btw when you update hekate you are not supposed to delete bootloader folder and your settings

Worked also for me, thx