CTCaer/hekate #1000: How

SpeakerGames, 2024-08-04:
How to launch linux or android?
[image]

urherenow, 2024-08-04:
There are tutorials everywhere. This isn’t a support forum and you could easily type that question into Google and found the answer.

SpeakerGames, 2024-08-07:
> There are tutorials everywhere. This isn’t a support forum and you could easily type that question into Google and found the answer.
I googled it 1000 times and haven't found an answer

CTCaer, 2024-08-07:
literally the first search results have switchroot wiki in them.
also here's not a chat/forum to ask how to installed/setup something.

closed 2024-08-07 by CTCaer