CTCaer/hekate #1001: Strange overlap when booting into emummc, sysmmc boots

AkTheCumDumpster, 2024-08-07:
something very weird is happening
so when i load into rcm, then hekate and launch my emummc everything works fine. but ive found that if im in emummc and i use the reboot to payload option then launch emummc again... it loads sysmmc and it connected online. i immediately turned airplane mode on and i checked it out and it has homebrew, but the weirdest thing is that, even though its clearly my sysmmc the system info in settings has the E at the end meaning it thinks its emummc. opening eshop shows that it couldnt connect just like on emu so i dunno whats going on. it only seems to happen after reboot to payload so far. if you need more info then let me know

CTCaer, 2024-08-08:
I don't know your setup but you probably have something rogue in it.
I suggest you to enable update reboot 2 payload in hekate Options to be sure.
If that payload is the latest but was replaced with an edited header, you might need to delete it (sd:/atmosphere/reboot_payload.bin).
And you should also check your .ini configs, since if you are using hekate to boot, it can support many emummc in a single sd.
(If your config does not use hekate to boot - aka no fss0 and payload - then here's the wrong repo to ask that.)

KinneticOvrdr1ve, 2024-08-08:
I followed the NH switch guide for homebrew, I didn’t really modify my setup and i checked my ini files but they seem fine, nothing out of the ordinary. Update reboot 2 payload is already on and i deleted the bin anyways just to make sure its clean after that happened
[image]
[image]

CTCaer, 2024-08-08:
What happens if you use the power menu to reboot? (Hold power until it shows the power menu.)

CTCaer, 2024-08-08:
"Restart" from the power menu reboots into hekate from that menu while in CFW. It doesn't power down "normally".
Something is wrong with your setup. I would suggest to refresh all files and remove any weird config from atmosphere, since it's what manages reboot method (easiest way is to delete atmosphere and reextract official release zip).

closed 2024-08-08 by CTCaer