CTCaer/hekate #1005: Gamepad not detected on Debian 12

Arctic-Circle-System, 2024-08-13:
I plugged it in as you normally would and started the gamepad mode. It says "Started HID emulation", but any games with controller support just don't detect it. It doesn't show up as a device in moltengamepad. Gamepadtool also doesn't work. I'm not sure what's going on.

CTCaer, 2024-08-24:
The gamepad exposed to OS is simple hid.
So it should work everywhere. Except if your OS makes the exposed interface as root only and it expects a udev rule to fix it.
You can check with `sudo evtest` and also that it shows up in lsusb and /dev/

Arctic-Circle-System, 2024-08-30:
@CTCaer Alright, I'll check that in a bit and let you know of the results. Thank you.

CTCaer, 2024-09-13:
@Arctic-Circle-System did you check?

CTCaer, 2024-09-24:
I'm assuming user fixed that at this point (probably a udev rule issue).

closed 2024-09-24 by CTCaer