CTCaer/hekate #1007: Booting into emummc after reboot to payload results in booting syscfw

aravind-n, 2024-08-15:
# Problem
When I launch hekate as a payload from my computer, I am able to launch my emummc correctly and the System Version string has E at the end. When I reboot to hekate and then launch emummc again however, I boot into sysCFW with the version string saying E at the end.

# Steps to reproduce
1. Enter RCM
2. Send Hekate 6.2.1 as payload
3. Launch Atmosphere F550 EmuMMC
4. Reboot to payload
5. Launch Atmosphere F550 EmuMMC
6. Should now be in Atmosphere F550 SYS

# Configuration:
* SysNAND firmware version: 18.1.0
* EmuMMC firmware version: 18.1.0
* Atmosphere version: 1.7.1
* Hekate version: 6.2.1

# Other notes
* All configuration files (eg: hekate_ipl.ini) sourced from switch hacks guide

aravind-n, 2024-08-15:
Seems to be an atmosphere issue and not a hekate issue. Closing this thread

closed 2024-08-15 by aravind-n

impeeza, 2024-08-15:
This is not a support site, please go to discord or GBATemp, and close this issue, issues are ONLY TO REPORT HEKATE'S BUGS