CTCaer/hekate #1009: Preinstall sanity check

non-bin, 2024-08-24:
Hi

I'm pretty late to the game I know, I just want to make sure I haven't missed anything because info is kinda sparse. I have a serial **XKW**7001123XXXX switch, on 18.1.0. I know I can't use fuzzy-glee so I bought an RP2040 Hwfly lite, and I'm planning to follow [this hardware guide](https://guide.nx-modchip.info) and [this software guide](https://switch.hacks.guide/user_guide/modchip/preparing_hekate/) probably using emuEMMC to be safe

Does anyone have any experience/recommendations/cautions/confessions-of-love for me?

Thanks :)

impeeza, 2024-08-24:
This is not a support site, please go to Discord or GBATemp

non-bin, 2024-08-26:
Sure thing, thanks :)

closed 2024-08-26 by non-bin