CTCaer/hekate #1012: Question: are third party joycons supported

FredHappyface, 2024-09-13:
Hi thanks so much for this awesome project

I thought this might be better suited to ask on discord but I cannot find a link 

Just wondering if 3rd party joycons are supported to navigate menus. It seems that when I use a 3rd party left joycon and an official right joycon, both are disabled? Is there anyway of using only the right joycon?

HOS and atmosphere both seem to work with these when docked (and wirelessly though I realise that's less relevant)

Thanks for your time :)

impeeza, 2024-09-13:
Only third party full compatible with rail connections could be supported, NO BLUETOOTH ON payload environment.  ping to CTCaer on discord.

CTCaer, 2024-09-13:
All 3rd party joycon should be supported, as long as they support wired connection via the rails, as stated above.
If yours is only BT then you can actually make the right joycon as primary.
edit `sd:/bootloader/nyx.ini` and change `jcforceright=0` to `jcforceright=1`

closed 2024-09-13 by CTCaer

FredHappyface, 2024-09-13:
Cheers for this, appreciated. I'll have a play around but yeah they are plugged in, and are not being picked up. I don't intend to use em over Bluetooth in hekate. Also happy to ping over discord if that's easier? Do you mind dropping me a link if that's your preference? 

Thanks again :)

CTCaer, 2024-09-13:
You can DM me. find my discord tag via retronx (I only allow DMs on mutual servers).
I will reopen the issue if it's a compatibility issue btw (if it really communicates via rails).

FredHappyface, 2024-09-13:
🤦‍♀️

Apologies! It turns out I'm an absolute fool and it works perfectly

Thanks for your patience on this!