CTCaer/hekate #1013: Black screen on boot with Hekate 6.2 or 6.2.1

Stef2024-sw, 2024-09-15:
Hello, I have black screen when I try to inject payload HEKATE 6.2 or 6.2.1 , no problem with HEKATE 6.1.1 or older.
New sd card tested, this is not the problem.
Even without sd card in the slot, when I inject payload 6.2 or 6.2.1, I received a message with some strange character and freeze after 10sec.
[image]
[image]

I think the problem is from my hardware but it's strange because working well till HEKATE 6.1.1

urherenow, 2024-09-15:
Just a shot in the dark, because my first (uneducated) guess is that you have a ram issue. Try to edit sd:/bootloader/nyx.ini and change bpmpclock=1 to bpmpclock=2 or 3.

Also, are you using a PC or Android device to inject, or are you using a dongle?

Stef2024-sw, 2024-09-15:
- I already try to put bpmclock=2 or 3 and it's the same black screen (for information with Hekate 6.1.1 I need to use bpmclock=2)
- I use PC to inject or webusb on macbook.

For information, here is the difference with HEKATE 6.1.1 injection (without sd card in the slot) : 
[image]
no problem like you see in comparison with picture in my first post.

I want to believe that the problem is the RAM but why do I have no problem with 6.1.1, I use it every day.

urherenow, 2024-09-15:
No idea, except an older version may use less or more of it. But also updates to memory card compatibility is also done between some versions. Is the SD info screen showing anything different between the two?

Stef2024-sw, 2024-09-15:
yes is different...

with 6.1.1 :  
[image]

with 6.2.1 : 
[image]

urherenow, 2024-09-15:
I'm talking about the SD info screen from within nyx. Are you not able to get into nyx? The screen you're showing is with no sd card or at least no nyx. Are you extracting the ENTIRE zip from releases to your card every time you change payloads? I hope you're not using some AIO pack.

Stef2024-sw, 2024-09-15:
I just test, when I put bootloader 6.2.1 directory in my SD card, never mind if I use bpmclock=2 or bpmclock=3... the screen remains black.

Stef2024-sw, 2024-09-15:
Just to be sure I make new tests:

Test 1
1- new sd card format fat32, add directory bootloader from hekate_ctcaer_6.1.1_Nyx_1.6.1.zip , nothing else.
2- I inject payload hekate 6.1.1
3- hekate start with this message : 
[image]
4- Power off and modify sd:/bootloader/nyx.ini with bpmpclock=3.
5- I inject payload hekate 6.1.1 and Hekate start normaly, no problem.

Test 2
1- new sd card format fat32, add directory bootloader from hekate_ctcaer_6.2.1_Nyx_1.6.3.zip , nothing else.
2- I inject payload hekate 6.2.1
3- Screen remains black, mandatory to power off.
4- I go to sd card directory bootloader and file nyx.ini don't exist.
5- I try to add nyx.ini manually with bpmpclock=3 but screen remains black.

Stef2024-sw, 2024-09-19:
On which discord can I talk about this problem?

urherenow, 2024-09-20:
RetroNX. You can also try your luck on gbatemp.

CTCaer, 2024-09-24:
That looks like a ram issue. In some hekate versions, the ram mapping changes and in 6.2.0 it did.
Reason why you have that issue there, since now it hits a broken area in your ram.

Stef2024-sw, 2024-09-24:
I suspected that the problem was with my console although currently I no longer use Hekate, I directly inject fusee.bin and I start directly on Atmosphere and it works correctly like that.
Is it possible that with a future version of Hekate I no longer have this problem?

CTCaer, 2024-09-28:
ram issue is ram issue and you will notice it if a game goes and uses the problematic area. So rearranging the memory addresses is a mute thing. It doesn't fix it, you just avoid it in the bootloader stage.

closed 2024-09-28 by CTCaer