CTCaer/hekate #1016: Skip L4T verification (Question)

AngelTorresGarcia, 2024-09-28:
I am experimenting with downgrading Android on OLED consoles but I have a problem when skipping the L4T verification. I am attaching a photo of my issue. Thank you very much for reading me and I hope for a response.
[image]

urherenow, 2024-09-28:
This isn't a help forum. Go to discord RetroNX.

AngelTorresGarcia, 2024-09-28:
Can you please sendme an invitation of the discord? 🙏🏻

urherenow, 2024-09-28:
You couldn't Google it?

https://discord.gg/erNC4FB

CTCaer, 2024-09-28:
That's not a verification thing. Either properly add the missing `boot_prefixes` path when you use `l4t=1` or use `payload=` if it uses the old coreboot method.
(`boot_prefixes` denotes the path where l4t bootstack files are located)

closed 2024-09-28 by CTCaer

AngelTorresGarcia, 2024-09-29:
> You couldn't Google it?
> 
> https://discord.gg/erNC4FB

You could just say no, you don't have to be a conceited idiot to answer, the information is appreciated but your way of answering is mentally retarded, by the way thanks and screw you.

urherenow, 2024-09-29:
Lol... idiot... says the idiot who can't Google a simple question 🤣

AngelTorresGarcia, 2024-09-29:
> Eso no es una cosa de verificación. O bien agregue correctamente lo que falta `boot_prefixes` ruta cuando usas `l4t=1` o usar `payload=` si utiliza el antiguo método coreboot. (`boot_prefixes` denota la ruta donde se encuentran los archivos de bootstack l4t)

Thanks, my situation is that I would like to try to make coreboot adapt to l4t or something similar, that's why I'm investigating what possible solutions work for me.