CTCaer/hekate #1020: Hekate error: "Missing LP0 (sleep) lib / Missing Minerva lib"

costantion, 2024-10-08:
Hi everyone,

I'm currently facing an issue with Hekate on my Nintendo Switch. When I try to boot, I get the following error message:

Missing LP0 (sleep) lib!
Missing Minerva lib!
Update bootloader folder!
Press any key...
[image]

What I’ve tried so far:
I’ve updated to the latest version of Hekate from the official GitHub page.

Question:
Does anyone have an idea what could be causing this issue? Is there anything I’m missing or something else I should try?

Any help would be greatly appreciated! Thanks in advance!

urherenow, 2024-10-08:
This isn't a support forum. It told you exactly what you need to do (update your bootloader folder). You must extract THE ENTIRE FOLDER to your card, overwriting all files. If you still get that error, it's still because it can't read those files, meaning your sd card (or reader) is damaged, or not formatted properly. If you've updated to FW 19, you also need to wait until a new version is released, but I don't believe this error has anything to do with that.

CTCaer, 2024-10-09:
You have a ram issue as it seems. hekate uses ram for various components and if it doesn't work, you can see errors like that.
And I can't do anything on hw issues.
[image]

closed 2024-10-09 by CTCaer

Link82dsd, 2024-10-19:
hello I have exactly the same error could you solve it?

Newbit13, 2024-12-18:
可能你的sd卡里的bootloader文件夹下有重复命名的文件，删掉其中一个就好。或者重别的地方重新拷贝别人的bootloader文件夹进去覆盖