CTCaer/hekate #1023: [BUG] failed to restore USER partition with split files

tldzyx, 2024-10-12:
- **hekate v6.2.1 & Nyx v1.6.3**
- **sd card format is FAT32**
- I backup my user partition by hekate, I get many files in sd card like **/backup/xxxxxxxx/partitions/USER.xx**
- then I move the files to path **/backup/xxxxxxxx/restore/partitions/USER.xx**
- now, I use hekate restore USER partition, it's failed

[image]

- at last, I merge the files into **/backup/xxxxxxxx/restore/partitions/USER**, then try again, it's working

[image]

review:

1. backup in FAT32 will auto split file √
2. restore partition not adapter the files ×
3. logs out screen Σ(⊙▽⊙"a

please fix them, thanks for you

CTCaer, 2024-10-12:
Try again with the latest version and show me the log again.
I removed the `Failed!` messages so now it's visible in its whole.

tldzyx, 2024-10-12:
> Try again with the latest version and show me the log again. I removed the `Failed!` messages so now it's visible in its whole.

- hekate v6.2.2 & Nyx v1.6.4

I'm try again with the latest version, this is the logs, looks like not found file

[image]

CTCaer, 2024-10-12:
I actually forgot. I don't allow split restore for partitions.
I guess I can in the future.

tldzyx, 2024-10-12:
> I actually forgot. I don't allow split restore for partitions. I guess I can in the future.

I believe. same like restore RAW GPP work very good.