CTCaer/hekate #1024: Sleep Mode Freeze

District7, 2024-10-13:
Hello updated to current hekate v6.2.2 & Nyx v1.6.4 and my OFW is firmware 19.0 but my CFW is 18.0.1 and I don't get the issue I had before but now after I am in CFW and sleep mode is activated I bet a frozen black screen have to force restart. is there something I can do to to fix this or need to wait for Atmosphere to be updated?

CTCaer, 2024-10-16:
Boot stock once for the updated sleep firmware to be dumped.
(Doesn't matter if it fails to boot without updating atmosphere, it will still be dumped.)

closed 2024-10-16 by CTCaer

District7, 2024-10-16:
> Boot stock once for the updated sleep firmware to be dumped. (Doesn't matter if it fails to boot without updating atmosphere, it will still be dumped.)

SO I boot Stock and get blue screen I let it sit then restarted but still same issue I also booted OFW then back to CFW and still sleep freeze happens, Under OFW I have not issue there as its all stock but my CFW is still freezing and needs hard reset to turn back on. I have turn off sleep mode for now but not an ideal way to run the switch. any help would be appreciated.

District7, 2024-10-16:
Atmosphere updated to support OFW 19 pre release updated and now have no issue using 19 firmware cross the board