CTCaer/hekate #1026: AutoRCM with SYSNAND 19.0.0 + EMUNAND 18.1.0 (Atmosphère 1.7.1) Not Working

WillsitoGG, 2024-10-17:
AutoRCM with SYSNAND 19.0.0 + EMUNAND 18.1.0 (Atmosphère 1.7.1) Not Working, when going to SYSNAND from Hekate it gives a HOS error in the screen vertical mode.

CTCaer, 2024-10-17:
>a HOS error

what's the error?

WillsitoGG, 2024-10-18:
> > a HOS error
> 
> what's the error?

If the Auto RCM is off, and I launch SYSNAND from latest Hekate version, the screen turns full yellow and the only way is to hold the power button for 12 seconds to turn it off.
[image]

If the Auto RCM is on, and I launch SYSNAND from latest Hekate version, the screen turns full black excep the following error:
HOS Panic Ocurred!
Color... Code: 06
Press any key
[image]

The only only change, before everything working as a charm, was pasting the latest hekate v6.2.2 & Nyx v1.6.4 files in the SD.

Hope it helps,

Best regards

Willsito

WillsitoGG, 2024-10-18:
I correct the title to one more accurate:
Booting, from Hekate v6.2.2 & Nyx v1.6.4, into SYSNAND 19.0.0 + EMUNAND 18.1.0 (Atmosphère 1.7.1) Not Working (either AutoRCM or not).

CTCaer, 2024-10-18:
Update atmosphere to 1.8.0. You can't boot 19.0.0 with 1.7.1. It doesn't have the necessary key generation.

closed 2024-10-18 by CTCaer

WillsitoGG, 2024-11-01:
But what do I do if I want to maintain the previous Atmosphere version, as it is far more stable compared to the last one, plus this last one has a lot of issues and bugs, for example, not being able to use forwarders?