CTCaer/hekate #1027: Please help optimize the payload tool “program_fuse_odm4_t210_t210b01.bin” (dram fuse burn)

Sean-China, 2024-10-19:
Please help optimize the payload tool “program_fuse_odm4_t210_t210b01.bin” (dram fuse burn) by adding two conditional branches. The specific requirements are as follows：
When dramid is 23, program fuse 33
When dramid is 28, program fuse 34
This will support reverting the memory from 8G back to 4G。

Currently, after upgrading my NS memory from 4G to 8G, I used the “program_fuse_odm4_t210_t210b01.bin” tool to change dramid to 28. Now，as I am trying to downgrade the memory from 8G back to 4G, but dramid remains 28, This is causing frequent errors in the CFW system。I hope you can help me，Thank you！

CTCaer, 2024-10-19:
**That tool is not for playing.**
If you wanted to play, you shouldn't have burn the fuse in the first place and just used a custom hekate.
Also your suggested IDs are wrong.

closed 2024-10-19 by CTCaer

Sean-China, 2024-10-19:
I did not change the dramid myself, but the person who helped me change the memory did。 Now I have changed the memory to "4GB_MICRON_MT53E512M32D1NP_046_WTB"，but because the dramid is still 28，it will cause many abnormal errors。 Please help me solve this problem。
Or can you provide the source code of "program_fuse_odm4_t210_t210b01.bin"，Thank you！

Sean-China, 2024-10-19:
I did not change the dramid myself, but the person who helped me change the
memory did😭。
Now I have changed the memory to "4GB_MICRON_MT53E512M32D1NP_046_WTB"，but
because the dramid is still 28，it will cause many abnormal errors。 Please
help me solve this problem。
Or can you provide the source code of
"program_fuse_odm4_t210_t210b01.bin"，Thank you！

Sean-China, 2024-10-19:
Sorry to trouble you，I know I made a stupid mistake，but now I really have
no choice but to ask you for help。Please help me think of a remedy when you
are free。

My machine has done the following：
1、The machine changed from Samsung 4G memory to Samsung 8G memory, and then
used the "program_fuse_odm4_t210_t210b01.bin" tool to change dramid to 28；
2、Recently，the memory was changed to Micron 4GB，but because dramid is still
28 (Samsung 8G)，it caused constant abnormal errors。