CTCaer/hekate #1028: [Enhancements] Uniform size for rawnand.bin and eMMC split files

tldzyx, 2024-10-19:
# Current Split Size

- the splited files `00~14` in path `/emuMMC/SD00/eMMC` max size is `4261412864 bytes`, equals `4064MB`, equals `2GB minus 32MB`
- the splited files `rawnand.bin.00~rawnand.bin.14` in path `/backup/xxxxxxxx/restore` max size is `2147483648 bytes`, equals `2048 MB`, equals `2GB`
- the splited files `rawnand.bin.00~rawnand.bin.14` in path `/backup/xxxxxxxx/restore/emummc` max size is `2147483648 bytes`, equals `2048 MB`, equals `2GB`

# why uniform size
in fact, the merge file at above three files group is the same file, uniform size let `backup / diff / transfer` could be easy and simple

# other questions
I noticed that nintendo split file in `***.nac` folder max size is `4294901760 bytes`, equas `4GB minus 64KB`, combine the above two, there are three different max split size for fat32 limit, let there use the same max size could becoming a good idea?

CTCaer, 2024-10-19:
[image]

closed 2024-10-19 by CTCaer

tldzyx, 2024-10-19:
> ![image](https://private-user-images.githubusercontent.com/3665130/378118974-2b3b507f-9971-4b24-a3e5-6b59b9109b59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzODE4MjYsIm5iZiI6MTcyOTM4MTUyNiwicGF0aCI6Ii8zNjY1MTMwLzM3ODExODk3NC0yYjNiNTA3Zi05OTcxLTRiMjQtYTNlNS02YjU5YjkxMDliNTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMTlUMjM0NTI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2YwMTdjMmRjYTU5ZDE1NjU1YTM2MDQxZDViYWVmYTgzZWRiM2IxNDExZmFkNDk3ZDYzMTJkMmZhOWE3NjcwMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.A5Z2aavzmI99DnRubM6SYlgblmublzX3juffxZllUO8)

why? I don't understand

impeeza, 2024-10-20:
> > 
> 
> why? I don't understand
Buddy, this is not a support site, please go to Discord or GBATemp.net.

your questions is a joke because you are comparing file size with file use on disk -the difference is about the size of cluster on disk-,  that's why your files have different sizes.

You can split files on the size you like, the most used ones are 2 GB or 4GB because the limits of the FAT32 format. but is mute which you use.

on the downloads section of the repo there are scripts to join the files and you end with a file of 32GB on size.

tldzyx, 2024-10-20:
> your questions is a joke because you are comparing file size with file use on disk -the difference is about the size of cluster on disk-, that's why your files have different sizes.
>
> You can split [...]

@impeeza I think you misunderstood. I'm not joke, maybe like discuss or request

- the split size was from hekate, not my choice
  - if create emuMMC will get `/emuMMC/SD00/eMMC/**`, each max size is 4064MB
  - if create backup will get `/backup/xxxxxxxx/rawnand.bin.**` or `/backup/xxxxxxxx/emummc/rawnand.bin.**` on fat32 sd card

- In my timeline
  - I found total size of `/emuMMC/SD00/eMMC` and `/backup/xxxxxxxx/rawnand.bin` is equals
  - I copy them on my computer, and try `cat 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 > rawnand.bin`, diff file sha256, exactly the same
  - I try `split -b 4064M -d rawnand.bin ''`, diff each file sha256, exactly the same
  - I try to convert sd file to sd partition by hekate, and I choose 58G for sd partition, so the sd card from exFAT to fat32
  - I create emu by sd partition, then I backup the emu, got `/backup/xxxxxxxx/emummc/rawnand.bin.**`, total was 58G
  - I guess and try, move the `/backup/xxxxxxxx/emummc/rawnand.bin.**` to `/emuMMC/SD00/eMMC/**`, and copy other like `BOOT0`/`BOOT1`/`file_based`, then choose emu to sd file, it's run success
  - so I conclude `/backup/xxxxxxxx/emummc/rawnand.bin.**` and `/emuMMC/SD00/eMMC/**` is the same file, just split size different
  - then I think, why there has different? why not use the same split size? I guess it's because fat32 limit, so I think the game file like `**.nac` has the same question, so I open the `Nintendo` folder and find and compute, it 4GB minus 64KB, so nintendo think it's a good split size?
  - I'm no expert. so I post issue at here, may it's can be uniform, just adjust the split arguments, or there has some reason let they are different I don't know
  - so, I got a photo, but I can't understand, so I hope get answer.
  - how ever, this issue was closed by maintainer, so I will not request it again, mybe I will get the answer at some place, or forget it.
  - I'm sorry I might be a little obsessive.

CTCaer, 2024-10-20:
The pic goes to you too @impeeza 

The sizes are very specific.
- backup size is 1G**i**B or 2G**i**B so someone can do a backup with only a 2GB or 4GB card without complicating the code.
  For exFAT with enough size it is dumped in its whole.
- emummc is constrained by fat32 limit and you are required to have all parts in it. So it's 4GiB - 32MiB.
  That 32MiB is for big sd card AUs
- everything for hekate is basically aligned to 4MB min for read/write performance reasons and that is a requirement on the split parts also.
 That's also why the file size matches filesystem size.
- nintendo is nintendo and it's irrelevant what it does (their 64KB alignment block has specific meaning though).
  What nintendo does has nothing to do with hekate basically and you can also not change it ofc.

The issue here is that I don't cater to OCDs for such things and that backup/transfer in "why uniform size" is not affected at all.
As for diffing, that must be an added argument without checking the specifics just for the shake of it, since almost the entirety of rawnand.bin has encrypted data.

In the end, the issue page is not for such things. You can ask in discord about these.