CTCaer/hekate #1029: No Warning Message when tapping Create emuMMC

MayorBryce, 2024-10-25:
For most operations that can delete data, such as SD card partitions, restoring NAND, etc., there is a 6s warning message before you can continue to prevent you from irreplaceably deleting your data. However, if you tap on emuMMC, Create emuMMC, and follow the steps, Hekate will begin overwriting the EmuMMC and copying the SysNAND, without a warning message to stop you.

Although you could argue no one uses this screen very often, I still think it's important for there to be a warning message. I accidentally fell for this while trying to help a friend out with a NAND backup, and was poking around in Hekate and accidentally tapped on this message. While I have a relatively recent NAND backup on hand, I can imagine an ignorant user wiping their EmuMMC my accident.

Masamune3210, 2024-10-27:
If you have to not only select the page but follow multiple steps before anything happens, one could argue the steps up to the action in and of themselves function as a warning. At what point does it stop being an accident and start being just someone being a bit more careless than they should be?

MayorBryce, 2024-10-27:
While I agree there was some carelessness on my part when I did this, I still think this is a noteworthy change.

* It’s inconsistent with operations of a similar purpose (SD card reformatting, NAND restoration, etc.)
* It could potentially prevent mistakes similar to the one I made
* It’s a relatively small feature to implement, without any downsides in regards to user functionality, coding complexity, etc.

Thank you.

CTCaer, 2024-10-28:
I don't consider emuMMC important data because it's always a copy of eMMC (sysmmc).
Plus that going to create, and then selecting a type, and then a partition to start is on user.
Especially when user can still abort the process like always by pressing both VOL buttons, before it reaches SYSTEM and USER partitions.

The only important thing is eMMC because if you lose or corrupt some data of it you are gonna have greater issues.
That the emuMMC restore option shows the warning is a side-effect because code is shared with eMMC restore.
As for partitioning, I find the user files that have nothing to do with switch or if eMMC backup, if existing, more important also.

I may add a rudimentary check at some point though and warn if it exists.