CTCaer/hekate #1030: [Question / Feature Request] Booting directly into UMS (Tools > USB Tools > SD Card)

jrlodge, 2024-10-27:
Apologies if I'm misunderstanding this, but I'm working on a way to remotely perform CFW updates (specifically Atmosphere updates when my Switch accidentally does a system update while running SysBot). I’m looking to boot directly into the UMS SD card reader mode so I can upload the updated CFW files without needing to navigate through the Hekate menus via the touchscreen, allowing me to do it solely via PC.

I've tried putting:

`[config]
ums=0`

in bootloader/hekate_ipl.ini on the Switch's SD card then reinjecting the Hekate.bin but it doesn't boot directly into the UMS mode like I expected. Is that the intended use of ums=0 in hekate_ipl.ini to set up automatic UMS mode? If not, do you know if there’s a way to set up Hekate to enter UMS mode from a PC without needing manual menu navigation?

I'd use FTPD, but once the CFW and Switch system version become mismatched, it can’t access the homebrew menu until the updated CFW files are on the SD card.

Any help is appreciated, thanks!

masagrator, 2024-10-27:
> Is that the intended use of ums=0 in hekate_ipl.ini to set up automatic UMS mode?

Such setting is not listed in readme.

jrlodge, 2024-10-27:
> I guess you are mixing ini file with Payload storage.

Correct, my bad. 

So would my goal of booting into UMS SD mode be achieved by editing 0x98 (ums[1]) and 0x97 (extra_cfg) in hekate_ctcaer_X.X.X.bin with something like HxD?

impeeza, 2024-10-27:
Or you can use Studious-Pancake overlay, it have a item to reboot the console to UMS, I use all the time.

I think UltraHand Overlay also have a option to that.

CTCaer, 2024-10-28:
Use studious pancake as said for reboot from HOS. Or edit the boot storage https://github.com/CTCaer/hekate?tab=readme-ov-file#payload-storage specific bytes if you want to inject from pc. So you can basically have one named hekate.bin and one ums.bin and choose what to inject.

closed 2024-10-28 by CTCaer