CTCaer/hekate #1031: Bricked after using USB Tool

Cris1245M, 2024-10-28:
I was trying to use the usb tool for the first time but didn't worked, so my console freezed on the UMS screen, so I forced shutdown, but after that my Switch didn't power on anymore, I already tried holding the power button but didn't worked, I had AutoRCM enabled and now my switch it's on black screen, RCM not detected and can't do anything with it

fennectech, 2024-10-28:
Did you hold for 15 seconds?   Try plugging into charger then holding power button for 15 seconds and then wait for 30 - 60 minutes.

Cris1245M, 2024-10-28:
> Did you hold for 15 seconds? Try plugging into charger then holding power button for 15 seconds and then wait for 30 - 60 minutes.

yes, didn't work

CTCaer, 2024-10-28:
Remove the cable from your switch. Restart your pc in case your pc's usb port hanged.
Reinsert and send a payload.
That assuming your battery is enough.

It's not possible to brick a switch like that.
Well except if your pc has issues with usb and send a weird voltage or something, which it would still not kill it but kill the ESD protection.
You should also edit `bootloader/nyx.ini` and set `bpmpclock=1` to `bpmpclock=2` or `bpmpclock=3`

closed 2024-10-28 by CTCaer