CTCaer/hekate #1033: Failed to match warmboot with fuses

Kingfadlan, 2024-10-31:
My cousins Mariko switch started suddenly showing this error a couple of months ago, and we've tried everything we knew. I already booting to sysMMC and it didn't work. We didn't update anything and we are new to the modding scene please help

CTCaer, 2024-10-31:
You need to boot into sysMMC (stock or cfw mode) with hekate.
In case that's what you tried, get a screenshot from hw & fuses info (in console info in hekate).

Kingfadlan, 2024-11-01:
Here is the screenshot. I have no idea which info you needed so it's the full screen
[image]

CTCaer, 2024-11-01:
When you said you didn't update anything, I thought you were talking about something else.
You have old hekate and you updated to 19.0.0.
So update hekate and atmosphere.

closed 2024-11-01 by CTCaer