CTCaer/hekate #1034: [Enhancement] 💡Allow direct value inputs for Partition Manager

MarsWonderBoy, 2024-11-01:
Title

As of now, when partitioning the SD card, the user must use the slider to their desired value. Add an option to simply tap at the value field to input a number directly.

I have a 512GB and partitioning for HOS, Linux, and Android. ...I've been fumbling between 61GB and 65GB when I only need 64GB to partition. 😭😭

[image]

HeikesFootSlave, 2024-11-01:
Nice Idea!

2x29 emuMMC!?
No matter which version I always set it to 12GB and never had any problems or issues! (it's enough for me and three profiles!

CTCaer, 2024-11-07:
The slider uses 700 pixels which means it allows for up to 700 selections.
If you find it difficult to select what you want, increase another (like linux) just to reduce max. And then reduce the other one back.
Or just use joycon mouse for even more precise movement.

On android though, what you see is not what you get (it just controls userdata partition and there are 6GB more of partitions), so dunno why you want to be precise.

closed 2024-11-07 by CTCaer