CTCaer/hekate #1036: NYX Error when tying to partition SD card.

ChicagGhost, 2024-11-04:
Hello,

I've been having an issue recently where my device has been freezing at the bootloader menu and every time I would restart it would continue to freeze, so I decided to start over and reinstall everything. 

I'm trying to partition my SD card so I can create an emuMMC. I keep running into an error from NYX saying the following, "NYX exception has occurred (LR 00000000)".

CTCaer, 2024-11-04:
Get a screenshot or photo from console info -> microsd.
Although the issue is probably RAM. (You can also try increasing bpmpclock in nyx.ini)

ChicagGhost, 2024-11-04:
Here is a picture of my SD card details, I did end up getting another error once I went into this.

(LR EB057576)

[image]

ChicagGhost, 2024-11-04:
Hello,

I tried what you suggested with increasing the bpmpclock,I went from 2 to 3 and that worked for. I was able to make do my partition, create emuMMC and it hasn't froze since once I increased that.

closed 2024-11-04 by ChicagGhost

CTCaer, 2024-11-04:
Even, better then. That means no hw issues.
(Btw the sd is legit and also there are no issues with sd connector)

HeikesFootSlave, 2024-11-04:
> Here is a picture of my SD card details, I did end up getting another error once I went into this.
> 
> (LR EB057576)
> 
> ![20241103_183627](https://private-user-images.githubusercontent.com/187224800/382617161-e45a8e9b-56e7-43d2-8a4c-86476fbd7959.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3MDY4OTIsIm5iZiI6MTczMDcwNjU5MiwicGF0aCI6Ii8xODcyMjQ4MDAvMzgyNjE3MTYxLWU0NWE4ZTliLTU2ZTctNDNkMi04YTRjLTg2NDc2ZmJkNzk1OS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNFQwNzQ5NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YTBkMWYyOWU0MGMwNjNhNDQ4MGFkZDQ0ZjBlZjllMDI4MWI3MjNjOWMzNTk5MDRlM2YwY2ZkNDY0ODIzYjA3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9._0PRjMqpZF_TMGsT7QLJyG5R4UMwRBrQP82ONWgcAv4)

Why are you using exFAT?
Shouldn't it be FAT32?

ChicagGhost, 2024-11-04:
> Why are you using exFAT? Shouldn't it be FAT32?

It is but this was more just to check the SD card itself the main issue was the freezing which has since been resolved.