CTCaer/hekate #1037: Not work for 19.0.1

SeFiROOTpitu, 2024-11-07:
I update ofw to 19.0.1 and then installed picofly, i run hekate fine but i cant create good emunand and not run in ofw, y get this message "unknow pkg1 versión, hos not supported" and then run in ofw, show Nintendo screen and then black screen

CostelaCNX, 2024-11-07:
Is not a hekate issue, is the Picofly install

SeFiROOTpitu, 2024-11-07:
I did the same with other ns mariko in 19.0.0 and working fine i created the emunand perfectly, but when update the ofw to 19.0.1 get the same message, only boot in emunand

Yahoo Mail: Busca, organiza, conquista

CTCaer, 2024-11-07:
Your package1 is corrupt in eMMC (sys). You need to fix it (it lives in BOOT0).

closed 2024-11-07 by CTCaer

SeFiROOTpitu, 2024-11-07:
I understand what happend here, if i disconect picofly the ns boot normaly, not emmc sys problem

CTCaer, 2024-11-07:
That's even worse actually. Because it means the signal gets corrupted by bad impedance.
Which also means that any use of HOS on eMMC (ofw/stock/cfw) can corrupt and software brick it by bad writes.
And you have no way of knowing if it already corrupted something.

SeFiROOTpitu, 2024-11-10:
Thank you, i Fix It, the problema was wire lenght