CTCaer/hekate #1039: [REQUEST] Changing the color of the regular text and icons

Noob0dooN, 2024-11-12:
Pretty simple i whould like a way to change the text and icons from white to a different color (Idealy black) and also maybe some brightness and saturation options for highlighted text

CTCaer, 2024-11-13:
Pretty simple... use nyx options and change color theme. And also use _hue icons.
Suggesting existing features is not the way to go.

closed 2024-11-13 by CTCaer

Noob0dooN, 2024-11-13:
> > Pretty simple... use nyx options and change color theme. And also use _hue icons. Suggesting existing features is not the way to go.
> 
> this is not what i meant, The "color theme" only sets the color for highlighted text Not regular text, and by icons i dont mean the launch icons i mean the smaller UI icons like the home button on the top. I want a option to set the default color of those icons (When not hovering). Hope this helps you understand what i mean

This is a quick mockup of what i want And ofc the top text should also be black im just too lazy
[image]

CTCaer, 2024-11-13:
Nope these will never get involved in the carnival.
But since you suggested "Idealy black" and that color can help with lighter backgrounds, maybe I will consider it.

If you want, create a new issue and properly ask that feature.
Aka option to set home screen menus icons and text to black for lighter backgrounds.

But have in mind my intention is for user to never use a lighter background because most look crap with UI elements (especially if not everything is tweaked).
For example backgrounds like that should not be chosen or they should be edited to something like that (I changed to look like it's a sunset):
[image]

Noob0dooN, 2024-11-13:
> Nope these will never get involved in the carnival. But since you suggested "Idealy black" and that color can help with lighter backgrounds, maybe I will consider it.
> 
> If you want, create a new issue and properly ask that feature. Aka option to set home screen menus icons and text to black for lighter backgrounds.
> 
> But have in mind my intention is for user to never use a lighter background because most look crap with UI elements (especially if not everything is tweaked). For example backgrounds like that should not be chosen or they should be edited to something like that (I changed to look like it's a sunset): ![bfj4q6hw8nn91](https://private-user-images.githubusercontent.com/3665130/385974239-a4c28abc-f1bd-4e2f-9c5a-b39119618376.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE1MzkxNjMsIm5iZiI6MTczMTUzODg2MywicGF0aCI6Ii8zNjY1MTMwLzM4NTk3NDIzOS1hNGMyOGFiYy1mMWJkLTRlMmYtOWM1YS1iMzkxMTk2MTgzNzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTNUMjMwMTAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjNlN2Q2MDE5NjVkMjE4YzI2ZGNkMzI5NmZlMWI4YmJkZDI1YjNjNDdlNGQ2YTllNWU2MzdhMzMwZWUzYTA5OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.keYTfLRlu3bsEbwUoKoh_9u7rKEBgbb7ZLPXqRQ3o0s)

Thanks for the info i will open a new issue now