CTCaer/hekate #1040: [Enhancement] Dark text for lighter backgrounds

Noob0dooN, 2024-11-13:
Add a dark text option for lighter backgrounds as it can be hard to destinguish like this example:
[image]

CTCaer, 2024-11-13:
Btw before I scrapped the idea of light theme I had these early 2019 alpha Nyx PoC builds:
[image]

And a later one when playing around with a more proper lighter theme implementation from scratch:
[image]

I guess I might search and find them at some point.

Noob0dooN, 2024-11-13:
> Btw before I scrapped the idea of light theme I had these early 2019 alpha Nyx PoC builds: [...]

If you do that whould be great and if you add it it you can put it in the nyx "Color Themes" at the top next to toggle background