CTCaer/hekate #1041: [Feature request] Moving data to new sd card using the docks inbuild usb ports

Noob0dooN, 2024-11-13:
Im not sure if this is possible
A tool for moving data (Partitions and files) to a differend sd card of same size or larger(If larger you will have to select partition sizes for each and they cannot go below the original value) using the docks build in usb port and a micro sd card adapter. and also using the non coved up part of the screen to show progress bar time remaining etc

CTCaer, 2024-11-14:
It's not and also completely out of scope.

Btw, instead of spamming the issues sections with random thought suggestions, it's better to contact me directly.
And actually using the issues search function is suggested (2 out of 3 issues created by you were already answered).

closed 2024-11-14 by CTCaer

Noob0dooN, 2024-11-14:
> It's not and also completely out of scope.
> 
> Btw, instead of spamming the issues sections with random thought suggestions, it's better to contact me directly. And actually using the issues search function is suggested (2 out of 3 issues created by you were already answered).

i kinda expected that and i'm sorry for the spam