CTCaer/hekate #1047: LR 4000E7E4 - hekate exception occurred

innocentDE, 2024-12-06:
Mod Chip installation with Picofly RP2040. Hang detected in LP0/Minerva.

First time this happened. Freshly installed. Modded already a few.

innocentDE, 2024-12-06:
- blue Screen when mod chip is removed
- boots into hekata (nyx is missing) when removing the SD Card
- emmc and sd card are find (sd card is legit and properly formatted)
- tools and console info from hekate are working

urherenow, 2024-12-07:
Did you search the closed issues for this error? It's almost always related to sd card/reader problems, file corruption, or you need to reflow your ram or apu. With an RPI, It could also be a wire that's a tad too long, or 1 of 2 spots that may need an extra 45ohm resistor (silicon lottery with that issue, I think).

pmnlla, 2024-12-09:
Blue screen is usually a bad NAND image or you need to reflow your board, iirc.

innocentDE, 2024-12-09:
> Blue screen is usually a bad OS image or you need to reflow your board, iirc.

Alright, will try to reflow. Thanks

innocentDE, 2025-04-18:
No, neither did i find a solution nor did I fix it.

leofrachet11, 2025-04-26:
I just found something really important and I was wondering if it works on your switch as well.

I'm was still trying to find why when there is no SD card installed, everything works fine, so I didn't install the SD card and it said "No SD Card" as I expected, but I found that when booting, if you hold both the volume down and volume up button at the same time, it bypasses this screen and boots into the OFW, and it worked! There were no issues or blue screens, proving that the issue isn't with the switch, so I was wondering if this would work on your one as well.

CTCaer, 2025-04-29:
That's probably a hw issue or your chip corrupts hekate when it's loaded.
Probably a reflow of the SoC will fix it.

closed 2025-04-29 by CTCaer