CTCaer/hekate #1050: Hekate far less responsive on Mariko units

souldbminersmwc, 2024-12-19:
Hello!
Hekate on mariko units is noticabily slower than on Erista units. Comparing Erista vs Mariko, Erista is faster and more responsive. My Mariko (OLED) is modded with a picofly.

masagrator, 2024-12-19:
This is not a bug. This is a result of not implementing ram training for Mariko units. Dev is fully aware of this.

Masamune3210, 2025-01-28:
That's basically what memory training is, its training on what is the best way to set the ram up. Since its part of very early init, I would imagine it's Hekate's job to handle it and it just isn't currently. There are only so many hours in the day and not all of that can be spent on dev work.

closed 2025-04-29 by CTCaer

souldbminersmwc, 2025-05-02:
Is this fixed or not?

impeeza, 2025-05-02:
> Is this fixed or not?

YES