CTCaer/hekate #1051: black screen in rcm mode after payload injection

nobru-rj-br, 2024-12-25:
I need help please!
I was playing Mario Party Jamboree (Atmosphere) and the Switch froze. After that, I can only turn on the console in OFW mode. (In OFW it works normally, I can play my original games).
When I put the console in RCM and inject, either from Android or via TegraRCMGui, without a micro SD inserted, the screen just goes black.

These are the two logs that appear within several attempts:
1)
No RCM device detected
RCM Device detected
Invoking TegraRcmSmash.exe with args : "C:\Users\-noBru-\Downloads\CNX.19.0.1-2\hekate_ctcaer_6.2.2 - Copia.bin"
TegraRcmSmash (32bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#6&30afba37&0&5#{aa0dbd45-3117-f331-5c49-76bf65225042}
RCM Device with id C08000190000000804532C6401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 109652, total size: 175868, total padded size: 176128)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected

2)
RCM Device detected
Invoking TegraRcmSmash.exe with args : "C:\Users\-noBru-\Downloads\CNX.19.0.1-2\hekate_ctcaer_6.2.2 - Copia.bin"
TegraRcmSmash (32bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#6&30afba37&0&5#{aa0dbd45-3117-f331-5c49-76bf65225042}
RCM Device with id C08000190000000804532C6401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 109652, total size: 175868, total padded size: 176128)...
Win32 error 31 happened trying to write payload buffer to RCM
 request!
Error while injecting payload (RC=-8)

urherenow, 2024-12-26:
dont create issues here if you're not even using hekate that came from here.

nobru-rj-br, 2024-12-26:
> dont create issues here if you're not even using hekate that came from here. AIO packs suck, and github isn't a help forum. Go to discord retronx or gbatemp.

I'm sorry. I downloaded Hekate from here and did the tests again. The problem persists:

No RCM device detected
RCM Device detected
Invoking TegraRcmSmash.exe with args : "C:\Users\-noBru-\Downloads\hekate_ctcaer_6.2.2_Nyx_1.6.4\hekate_ctcaer_6.2.2.bin"
TegraRcmSmash (32bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#6&30afba37&0&5#{aa0dbd45-3117-f331-5c49-76bf65225042}
RCM Device with id C08000190000000804532C6401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 109652, total size: 175868, total padded size: 176128)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected

urherenow, 2024-12-26:
1) Not a help forum, as already stated 
2) Downloading from here does not replace the configuration files you installed with the AIO pack, and who knows what system modules are included that you don’t even know about.

CTCaer, 2025-04-29:
I can't help with such HW issues.

closed 2025-04-29 by CTCaer