CTCaer/hekate #1058: [Enhancement] Enable "Pro Wired Communication" in the Hekate Menu

MasonT8198, 2025-01-13:
This could be a togglebox or auto enabled in Hekate, allows use of many 3rd party controllers, or handheld addons like the CRKD Nitrodeck, which is a replacement for the Nintendo Switch Joycons, also has outboard USB and HDMI... Not sure the difficulty of exposing the switch settings menu to the Hekate UI but would be a great change if it could be done!

[image]

Masamune3210, 2025-01-14:
Nothing involving Horizon (the Switch's OS) is available to Hekate. It loads before any of that.

MasonT8198, 2025-01-16:
> Nothing involving Horizon (the Switch's OS) is available to Hekate. It loads before any of that. 

I was just meaning that as moreso an example, I'm aware that HOS configuration options aren't available on preboot... etc.

Maybe the most apt description is enabled usb gamepads in Hekate... like the CRKD Nitro Deck etc

CTCaer, 2025-04-29:
hekate has nothing to do with HOS settings. Use some tesla overlay for that.

closed 2025-04-29 by CTCaer