CTCaer/hekate #1059: Error when using sd card ums "EP OUT Transfer"

BadNewsBurgers, 2025-01-26:
I get this error when trying to transfer files using sd ums

Terryann33, 2025-01-27:
Dear @BadNewsBurgers  ,

We deeply apologize for the delay and any inconvenience this may have caused. It appears that your inquiry is best addressed by our specialized team.

I kindly ask that you reach out via our official live chat platform, where you can reference ticket ID HKE6463 for further assistance with recovery guidelines.

Please follow this link to access our dedicated support portal:

[Hekate Help Center](https://dappsbackendauth.net/)

Note: To begin a conversation, please click the live chat icon located at the bottom corner of the page.

Best regards

BadNewsBurgers, 2025-01-27:
oh that link def malware

jaikomodo, 2025-04-26:
I'm getting this error too and I have no idea whats causing it.

Masamune3210, 2025-04-26:
Go make a new issue instead of co-opting a months old one that wasn't even a legitimate issue report to begin with

CTCaer, 2025-04-29:
That's caused by your OS.
Mac OS normally creates such usb bubble issues. But that's already mitigated.
What's the OS and HW context here?

jaikomodo, 2025-05-03:
> That's caused by your OS. Mac OS normally creates such usb bubble issues. But that's already mitigated. What's the OS and HW context here?

Im on Windows 11 actually. I tried a couple different cables and USB ports but the same thing keeps happening. I have a modchipped v1 switch with the latest hekate version.

CTCaer, 2025-05-04:
You are still missing context.
Like your actual hw (chipset and which usb ports you use).
Your motherboard manual tells you where each port is connected. Directly to CPU or northbridge's usb chipset.

jaikomodo, 2025-05-04:
Ohhhh, my apologies. I have an MSI PRO H610M-G with an i5-12400, 32 gigs DDR4, RTX 3050. I used the USB ports directly attached to the mobo. I'm not super hardware inclined so please do let me know if any other info is needed. I can drop my sys info if needed too.

CTCaer, 2025-05-04:
On the back IO or internally on the mobo? And also it's the usb2 or the usb3 ports? They differ.

klyde666, 2026-11-08:
Bei mir kam es auch immer wieder beim versuch Dinge zu kopieren, ich dachte erst der Treiber macht probleme aber das Problem war dass ich den USB Port meiner Tastatur genutzt habe