CTCaer/hekate #1063: I don't understand

Andrew15141, 2025-01-30:
I've download all the things that needed to be downloaded I found on the website my switch is 10.9.1 I've extract all the files that needed to be,  I don't understand what else I need to do I've never modded a switch but spent hours on figuring it out if someone can help me out and lmk why I'm getting these errors and how to fix that that would be great

[image]

[image]

urherenow, 2025-01-30:
You didn’t properly extract the zip files. Notice you don’t have a folder called “bootloader”. That’s because the way you extractd your stuff, you’ll see in inside if the hekate_ctcaer folder… grab EVERYTHING from inside ALL of those folders and move the contents to the root of your sd.

Andrew15141, 2025-01-30:
[image]

So it should look like this

urherenow, 2025-01-30:
Except the joiner scripts don’t need to be there. Those are for your PC.

Andrew15141, 2025-01-30:
Tyvm I might run into more problems since I'm not that smart in this stuff but I'll comment if I do

urherenow, 2025-01-30:
Please don’t. GitHub is for collaboration and reporting bugs. It is not a help forum. If you need help, go to retronx on discord, or gbatemp.

Andrew15141, 2025-01-30:
Ooo okay my bad will do

CTCaer, 2025-04-29:
I'm assuming user fixed the issue.

closed 2025-04-29 by CTCaer