CTCaer/hekate #1065: Drifting in Hekate and Android/Linux on Switch Lite

supersillysammy, 2025-02-17:
Hi, I've seen that someone made an issue about this in the past and that it was said to have been fixed, but I'm still experiencing this issue. Whenever I'm in Hekate or using Android and Linux my controller drifts occasionally, but it doesn't drift in HOS. I've done the calibration in both EmuNAND and SysNAND and dumped the joy-con pairing again and it's still happening.

pinkyp1234, 2025-04-27:
Hello, let me try to answer. English is not my main language. 

I had the same issue and I did able to fix it. Basicly, I follow the tutorial here by leloc900 https://gbatemp.net/threads/switchroot-android-11-lite-recalibrate-after-replace-joystick.658027/

Basicly, you only need to edit the number on switch.cal that located on switchroot folder. The tutorial have a link of a document that can help you determine the number you need for the switch.cal. don't forget to reboot your switch. You can use the gamepad tester via your android OS (lineage), just move your joystick when you open the web, the web will recognize it. 

I did follow the tutorial and changed the number I got from the document. I hope this help.

pinkyp1234, 2025-04-27:
Oh, one more thing. Mine is switch lite.

CTCaer, 2025-04-29:
You can change the dumped calibration by editing the file.
You should ask in the relevant discords or follow that guide I guess.

closed 2025-04-29 by CTCaer