CTCaer/hekate #1081: A new OFW Version has appeared

verFelixt, 2025-04-30:
20.0.0 is out. Do you already work on a new Version of Hekate? I accidentally updated my OFW and now my CFW is not working anymore.

Ninjin499, 2025-04-30:
Same thing for me. Lowkey ruined my evening

Cuphead74832, 2025-05-01:
But like, does hekate boot atleast?? Without the atmosphere or the CFW?

verFelixt, 2025-05-01:
[image]

[image]

It is booting yes.
But there is no chance to Launch the SysMMC CFW and the Stock SysMMC as well.

igoraresti, 2025-05-01:
Same here. Its not loading. It says error reading package1.

eduardez, 2025-05-01:
Same here, HOS version not supported on Hekate v6.2.2

[image]

Correct me if I am wrong, but maybe updating [hos.c](https://github.com/CTCaer/hekate/blob/master/bootloader/hos/hos.c), [hos.h](https://github.com/CTCaer/hekate/blob/master/bootloader/hos/hos.h), [pkg1.c](https://github.com/CTCaer/hekate/blob/master/bootloader/hos/pkg1.c) and [pkg2.c](https://github.com/CTCaer/hekate/blob/master/bootloader/hos/pkg2.c) is enough?

urherenow, 2025-05-01:
Ok, you’re wrong. There is no compatible package3 for hekate to load, so you’re not going to get anywhere. You all need to sit back and wait for Atmosphere to get updated, before you can then wait for hekate to get updated (or… simply launch fusee instead, still after it’s updated).

Cuphead74832, 2025-05-02:
> [image]
> 
> [image]
> 
> It is booting yes. But there is no chance to Launch the SysMMC CFW and the Stock SysMMC as well.

Does this mean a backup can be made of the eMMC without the CFW for later restoring into an EmuMMC? I need to know as im thinking of something.

RayliGeldman, 2025-05-03:
I don't want to be impatient but can it take a long time or will it go quite quickly because I think 20.0.0 is a pretty big update?

IkarugaOne, 2025-05-03:
I updated to 20.0.1 through Daybreak, it's great that you can download firmwares through the All in One Updater that aren't supported yet. Any way to roll back the firmware? I guess not, eh? So I can just create a new SD Card with the old firmware and all and use that instead and wait...

impeeza, 2025-05-04:
> I updated to 20.0.1 through Daybreak, it's great that you can download firmwares through the All in One Updater that aren't supported yet. Any way to roll back the firmware? I guess not, eh? So I can just create a new SD Card with the old firmware and all and use that instead and wait...

This is no a support site.  the only way so far is using the QA version of Atmosphère on Reswitched Discord, but please go to a support site for help.

next time do not upgrade any firmware until is supported by scene.

CTCaer, 2025-05-04:
As always hekate will not be released without atmosphere release that supports 20.0.0.
I want no use coming back here with blue/yellow panic screens.

closed 2025-05-04 by CTCaer