CTCaer/hekate #1082: [Bug] and [request] Costum background

Woody-NX, 2025-04-30:
Hi CTCaer,
I want to use a costum background in Hekate. But now I have the problem that as soon as I boot into the launch menu the costum background appears, but when I go home then back to launch it is grey/black again. Is there any way to fix this? Also, would it be possible to have a different image for each page? For example, for tools and console info?

Vampitech, 2025-05-01:
> Hi CTCaer, I want to use a costum background in Hekate. But now I have the problem that as soon as I boot into the launch menu the costum background appears, but when I go home then back to launch [...]

You need to dig in the source code to achieve this. There's no other way.

CTCaer, 2025-05-04:
That's done on purpose. Please search closed issues before opening one.

closed 2025-05-04 by CTCaer