CTCaer/hekate #1083: Please Add HOS 20

hadibh17, 2025-05-01:
Firmware 20 needs HOS Please Fix That
Thank You!

Aruno4, 2025-05-01:
> Firmware 20 needs HOS Please Fix That Thank You!

How about you freaking wait.

hadibh17, 2025-05-01:
I just want to fix my problem and i dont know how if you know please help me, thanks

ZomkaDEV, 2025-05-01:
Duplicate of #1081 

Why can't people be patient... no one is maintaining Hekate (or Atmosphere) as a job...

Brixxian, 2025-05-03:
> [@hadibh17](https://github.com/hadibh17) It's funny you say "my problem". Well, here goes some news for you. Many people like you have the same issue because, they updated when main rule of the use of this stuff is (guess what?) NOT UPDATING!!. Creating an issue here will not solve "your problem" any faster. Just as you had the guts to upgrade, now have the guts to wait until everything is in order so you can have the right tools, and that takes time. For now, there's nothing you or any of us can do about "your problem" right now. So let other people do their work.

i am getting reddit vibes from this

MrRagtime, 2025-05-03:
> i am getting reddit vibes from this

nah I'm getting gmod rp vibes

CTCaer, 2025-05-04:
@urherenow stop talking bs btw. hekate is a bootloader.
Actually, I think it's time to cut the authoritative reply noise.

As for the 20.0.0, as always it will be released when atmosphere gets released. I don't want blue/yellow screen issues opening here for no reason.

closed 2025-05-04 by CTCaer