CTCaer/hekate #1086: HOS 20.0.0 support

CTCaer, 2025-05-04:
This is a testing archive that can be paired with atmosphere 1.9.0 testing one (20.0.0 support).
If you don't know what you are doing, it's generally better to wait for the proper releases.

This will not be the final release.

[hekate_20.0.0_support_test.zip](https://github.com/user-attachments/files/20029984/hekate_20.0.0_support_test.zip)

SupremeBoiii, 2025-05-05:
Could we get the source for that?

caojiang314, 2025-05-05:
THINK YOU

Cuphead74832, 2025-05-05:
Hey i have a question! If i use normal hekate build with HOS 19 support. Cud i use the bootloader to just backup my entire eMMC for later using this backup in the future in the emuMMC instead of booting a payload to CFW even if the HOS IS unsupported??
(Also if theres a better site to ask please tell me)

Zejasght88, 2025-05-05:
Can someone help me get the URL for atmosphere 1.9.0? But i working with the version oficial with this URL https://github.com/olliz0r/sys-botbase/releases.

impeeza, 2025-05-05:
> Hey i have a question! If i use normal hekate build with HOS 19 support. Cud i use the bootloader to just backup my entire eMMC for later using this backup in the future in the emuMMC instead of booting a payload to CFW even if the HOS IS unsupported?? Just in case when the times comes and Nintendo decides to pull another Big OS update after this one. (Also if theres a better site to ask please tell me)

Yes you can

> Can someone help me get the URL for atmosphere 1.9.0? But i working with the version oficial with this URL https://github.com/olliz0r/sys-botbase/releases.

Is on the Atmosphères official repository on the issue about FW 20 support.

Thorhax, 2025-05-06:
Are modchips booting normally with the hekate update, or do we still need to wait for modchip firmware updates?

matthewperiut, 2025-05-06:
Modchips work.

Cuphead74832, 2025-05-06:
> Yes you can
>
>
> Is on the Atmosphères official repository on the issue about FW 20 support.

Thank you, i plan to backup my entire eMMC before transfering then use It after transfering in CFW by flashing the backup emuMMC.

ELY3M, 2025-05-08:
I have OLED with picofly modchip.  I updated it to 20.0.1 and I have my atmosphere fork working fine now.

ELY3M, 2025-05-08:
you need to get latest test build.

LordFransie, 2025-05-08:
I'm on an OLED with a modchip, when I boot to stock firmware with this I get a blue screen.

CTCaer, 2025-05-09:
hekate was released properly so update to latest.
(btw blue panic screen means new hekate and old atmosphere)