CTCaer/hekate #1090: Device key, secure boot key are FFFFFF

angeluskun, 2025-05-11:
[image]
Since I updated to 20.0.1 and use old Hekate version prior update that info is blank idk why, anyways my switch is ok (8

masagrator, 2025-05-11:
That's normal. It happens on every unit with modchip. Dunno why did you think it was different before, but you are wrong.

CTCaer, 2025-05-11:
Reading of that is locked out by Nvidia's design:
- On T210 (erista) unpatched: you can read it if you use the RCM exploit
- On T210 (erista) patched: boot via hekate and then reboot via R2P (warmboot exploit) to read it
- On T210B01 (mariko): you can't read it normally. use another tool (can be used for the above too).

I guess I should change the FF array into a "locked out".

closed 2025-05-11 by CTCaer