CTCaer/hekate #1092: offers

arseniy2043, 2025-05-12:
add the ability to open .nro sentences directly in hekate without launching atmosphere

impeeza, 2025-05-12:
Humm I think is way beyond the scope of Hekate,  NRO are software for HOS, and Hekate runs before any OS is loaded, and even on another processor.

impeeza, 2025-05-13:
Where in the tools section of Hekate there is such thing as NRO lis?

tools have:

[image]

and

[image]

I think you are confusing Payloads with NROs, THEY ARE COMPLETELY DIFFERENT PROGRAMS.

arseniy2043, 2025-05-13:
yes, there is no such thing in tools, because I just suggested such a location. I think it is quite possible, because .nro will be opened in a special payload

impeeza, 2025-05-14:
> yes, there is no such thing in tools, because I just suggested such a location. I think it is quite possible, because .nro will be opened in a special payload

Something seems is not to clear to you is:

Hekate run on a computer separated from the computer where Nintendo Switch OS (Horizont - HOS) run, yes, the console have two CPUs on the same chip. Is like you ask for run Windows software directly on a IOS will no happen.  Hekate initialize the console, then loads the third stage of boot and allows the HOS to load, for run NRO you need the HOS.

CTCaer, 2025-05-24:
What even...

closed 2025-05-24 by CTCaer