CTCaer/hekate #1095: Not working touchscreen

cloge01, 2025-05-18:
i upgrade hos 20.0.1 then I chipped my nsw lite and download ultranx and this happens , i tried unchip , load fullstock, i checked everything inside switch and in switch all fine, tried replase sd card, tried to replace card slot and it not work.

cloge01, 2025-05-18:
[image]

[image]

impeeza, 2025-05-18:
THIS SAME ISSUE WAS CLOSED ON ATMOSPHÈRE REPO, YOU ARE USING A MISLEADING CAPTION, YOU TELL «EVERYTHING WORKS FINE ON HEKATE» BUT YOU SHOW A PICTURE OF «CALIBRATION FAILED»  SO YOU MESSEDUP SOMETHING ON YOUR HARDWARE, AS TOLD BEFORE IS NOT A HEKATE NOR ATMOSPHÈRE ISSUE, REPAIR YOU HARDWARE.

cloge01, 2025-05-18:
> THIS SAME ISSUE WAS CLOSED ON ATMOSPHÈRE REPO, YOU ARE USING A MISLEADING CAPTION, YOU TELL «EVERYTHING WORKS FINE ON HEKATE» BUT YOU SHOW A PICTURE OF «CALIBRATION FAILED» SO YOU MESSEDUP SOMETHING ON YOUR HARDWARE, AS TOLD BEFORE IS NOT A HEKATE NOR ATMOSPHÈRE ISSUE, REPAIR YOU HARDWARE.

touch work but not calibrating idk how it work

closed 2025-05-24 by CTCaer