CTCaer/hekate #1097: Issue with UMS

superxoi, 2025-05-22:
I have 3 modded switches. The lite and oled can connect to PC via UMS. But the v2 can not. I have cheked hardware's side  and 2 lines D+ D- are presented on the usb port. At this point i don't know why it has this issue. If anyone encounter this issue before please help me.
Thanks in advance.

impeeza, 2025-05-22:
First, THIS IS NOT A SUPPORT SITE, THIS SPACE IS FOR REPORT HEKATE BUGS AND ISSUES ONLY.

Second,  try the Gamepad tool of Nyx (nyx is the GUI of Hekate)  on Tools -> USB Tools -> USB Input Devices -> Gamepad.  to test if another type of device is recognized by your pc. may be a drivers isse.

if that do not work go to Discord or GBATemp for help.

CTCaer, 2025-05-24:
Assuming that all 3 were tested with the same port on PC and the same cable, the issue is either your typec port or the esd chips are fried or your soc's usb logic is fried.
Nothing I can do for hw issues.

closed 2025-05-24 by CTCaer