CTCaer/hekate #1099: Booting in to Hekate Crashes Unpatched Switch

Ikenowski, 2025-05-26:
For whatever reason, after updating to 19.0.1 and installing all the new software for it, I am unable to boot into Hekate and it crashes my system. I've tried using RCM and rebooting into Hekate from Atmoshpere and neither work. I did get an error once that Nyx was not installed, other than that one time, the switch black screens and I have to force reset. I dragged and dropped everything from the zip file into the root of my SD card and used hekate_ctcaer_6.2.2.bin as the payload in my injector.

Ikenowski, 2025-05-28:
Understanding that this not a support site, I appreciate your attempt to help.

CTCaer, 2025-05-29:
Yeah this is not a support forum but anyway.
If you have a fake sd card, replace it. If files are corrupt replace them, except if the whole FS is, where you need a chkdsk or format.
If you updated by removing bootloader folder, you must never do that. hekate is designed to be updated by retaining the configs.

That also means that if you had a bpmpclock of 2 or 3 before, you lost it and you need to edit it back.