CTCaer/hekate #1101: Hello, could you help me?

degrahne, 2025-05-29:
I tried to use it on a new SD card, but it gives an error in pkg1 <'~p^8z'>. I formatted the memory and repeated the process, but it didn't work. Does anyone have an idea on how to fix it?

ViRb3, 2025-05-29:
Check if your SD card is not counterfeit: #366

Apart from that, make sure you're on latest Hekate and Atmosphere version.

RayliGeldman, 2025-05-29:
i have the same problem you probly updated to the latest version 20.1.0 and atmos and hekate dont have the newest version from it

ViRb3, 2025-05-29:
Both Atmosphere and Hekate were updated for 20.1.0. Just make sure you grab their latest versions as of right now.

RayliGeldman, 2025-05-29:
when i put atmosphere and hekate the latest version it still doesn't work, i thought that worked before the update 20.0.1, not for 20.1.0?

Masamune3210, 2025-05-29:
Sigpatches are considered piracy.

RayliGeldman, 2025-05-29:
Really, you need them to play games. I have my own game copies lying around but I can't put them on otherwise. Copied from the game cards

Masamune3210, 2025-05-29:
They aren't designed to be "put on", that's why the system has to be patched to allow it to happen. I don't like it either, but this is how it has to be. Sigpatches are piracy and projects can't be seen as even slightly promoting piracy or allowing it or risk getting shut down.

CTCaer, 2025-06-02:
piracy is not supported

closed 2025-06-02 by CTCaer