CTCaer/hekate #1104: Fatal error

Zejasght88, 2025-05-31:
I'm update heckate, but can't excecute de sysnand with fatal error

[image]

impeeza, 2025-05-31:
As the screen SHOWS, it is a ATMOSPHÈRE ERROR nothing to do with Hekate.

PLEASE READ, THE GUIDES ABOUT UPGRADING BY THE LETTER.

YOU MUST BACKUP AND REMOVE THE SDMC:/atmosphere/contentes folder.

Then reinstall the SYS-Modules, mods, cheats and other content MANUALLY WITH VERSIONS SUPPORTED BY YOUR CURRENT FIRMWARE. at this case is a SYSMODULE called Mission Controll outdated.

PLEASE CLOSE THIS ISSUE.

Zejasght88, 2025-06-01:
Wow, thanks for you help!!! you are the best

Zejasght88, 2025-06-01:
How close de Issue??