CTCaer/hekate #1113: CAN'T inject hekate payload on my Switch V1

odilonhg, 2025-06-28:
Hello, i've recently buy a Switch from 2017 to use it with Atmosphere but the Switch can't runs payloads...
In TegraRMC the loading of injection never stop.

My Switch is in the latest version 20.1.5, is it the reason ?
If it is, how can I continue ?
Nobody talks about the latest versions of the Switch (20.XX.XX)...

Thanks for your time !

AlexTech01, 2025-06-28:
Did you check on [ismyswitchpatched.com](https://ismyswitchpatched.com) to make sure it's not a patched V1 Switch? Some Switches with model number `HAC-001` are still patched even though they're V1

BasGame1, 2025-07-18:
Hi, I have the same problem, but on ismyswitchpatched.com doesnt appear my prefix

CTCaer, 2025-08-17:
Not an issue with hekate, so closing.