CTCaer/hekate #1133: pc doesn't detect usb mass storage

kitschxx, 2025-08-29:
i've been trying to use hekate's usb mass storage option to transfer files without removing sd card, but for some reason it is only stuck on waiting for connection, but windows detects a sound but nothing shows up on device manager. i tried editing the nyx.ini but still no luck, i also tried different cables but no luck. but i know it is not a hardware issue since dbinstaller works, but hekate just simply won't.

Leerz, 2025-08-29:
unplug your cable
shutdown switch
turn on switch and launch hekate
launch usb ums
plug in your usb cable

there are instances wherein the usb profile doesn't change and it affects mounting

CTCaer, 2025-08-29:
Did you replace the usb mass storage with another driver by mistake maybe?
many users do that with libusbk for example